ajfisher/node-pixel #291: Error installing on M1

itsflores, 2022-02-20:
Hi everyone! I've been trying to install node-pixel for a few hours and I honestly don't know what do try next.

Any help would be appreciated. Here are my logs

```
code 1
npm ERR! path /Users/omarflores/Developer/openFrameworks/apps/myApps/lumens/server/node_modules/firmata/node_modules/serialport
npm ERR! command failed
npm ERR! command sh -c prebuild-install || node-gyp rebuild
npm ERR! CXX(target) Release/obj.target/serialport/src/serialport.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.4.1
npm ERR! gyp info using node@17.5.0 | darwin | arm64
npm ERR! gyp info find Python using Python version 3.9.10 found at "/opt/homebrew/opt/python@3.9/bin/python3.9"
npm ERR! gyp info spawn /opt/homebrew/opt/python@3.9/bin/python3.9
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/opt/homebrew/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/omarflores/Developer/openFrameworks/apps/myApps/lumens/server/node_modules/firmata/node_modules/serialport/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/opt/homebrew/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/omarflores/Library/Caches/node-gyp/17.5.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/opt/homebrew/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/omarflores/Library/Caches/node-gyp/17.5.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/omarflores/Developer/openFrameworks/apps/myApps/lumens/server/node_modules/firmata/node_modules/serialport',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/serialport.cpp:41:48: error: too few arguments to function call, single argument 'context' was not specified
npm ERR!   v8::String::Utf8Value path(info[0]->ToString());
npm ERR!                              ~~~~~~~~~~~~~~~~~ ^
npm ERR! /Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/v8-value.h:360:44: note: 'ToString' declared here
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<String> ToString(
npm ERR!                                            ^
npm ERR! ../src/serialport.cpp:41:25: error: no matching constructor for initialization of 'v8::String::Utf8Value'
npm ERR!   v8::String::Utf8Value path(info[0]->ToString());
npm ERR!                         ^    ~~~~~~~~~~~~~~~~~~~
npm ERR! /Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/v8-primitive.h:507:5: note: candidate constructor not viable: no known conversion from 'MaybeLocal<v8::String>' to 'const v8::String::Utf8Value' for 1st argument
npm ERR!     Utf8Value(const Utf8Value&) = delete;
npm ERR!     ^
npm ERR! /Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/v8-primitive.h:500:5: note: candidate constructor not viable: requires 2 arguments, but 1 was provided
npm ERR!     Utf8Value(Isolate* isolate, Local<v8::Value> obj);
npm ERR!     ^
npm ERR! ../src/serialport.cpp:48:53: error: too few arguments to function call, single argument 'context' was not specified
npm ERR!   v8::Local<v8::Object> options = info[1]->ToObject();
npm ERR!                                   ~~~~~~~~~~~~~~~~~ ^
npm ERR! /Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/v8-value.h:372:44: note: 'ToObject' declared here
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
npm ERR!                                            ^
npm ERR! ../src/serialport.cpp:48:25: error: no viable conversion from 'MaybeLocal<v8::Object>' to 'v8::Local<v8::Object>'
npm ERR!   v8::Local<v8::Object> options = info[1]->ToObject();
npm ERR!                         ^         ~~~~~~~~~~~~~~~~~~~
npm ERR! /Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/v8-local-handle.h:157:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'MaybeLocal<v8::Object>' to 'const v8::Local<v8::Object> &' for 1st argument
npm ERR! class Local {
npm ERR!       ^
npm ERR! /Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/v8-local-handle.h:157:7: note: candidate constructor (the implicit move constructor) not viable: no known conversion from 'MaybeLocal<v8::Object>' to 'v8::Local<v8::Object> &&' for 1st argument
npm ERR! /Users/omarflores/Library/Caches/node-gyp/17.5.0/include/node/v8-local-handle.h:161:13: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
[... 138 lines cut ...]
npm ERR!                        ~~~~~~~~~~^
npm ERR! ../src/serialport.cpp:413:17: error: no member named 'Handle' in namespace 'v8'
npm ERR!   void init(v8::Handle<v8::Object> target) {
npm ERR!             ~~~~^
npm ERR! ../src/serialport.cpp:413:36: error: use of undeclared identifier 'target'
npm ERR!   void init(v8::Handle<v8::Object> target) {
npm ERR!                                    ^
npm ERR! ../src/serialport.cpp:413:43: error: expected ';' after top level declarator
npm ERR!   void init(v8::Handle<v8::Object> target) {
npm ERR!                                           ^
npm ERR!                                           ;
npm ERR! 7 warnings and 16 errors generated.
npm ERR! make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
npm ERR! gyp ERR! build error
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/opt/homebrew/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:526:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Darwin 20.6.0
npm ERR! gyp ERR! command "/opt/homebrew/Cellar/node/17.5.0/bin/node" "/opt/homebrew/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /Users/omarflores/Developer/openFrameworks/apps/myApps/lumens/server/node_modules/firmata/node_modules/serialport
npm ERR! gyp ERR! node -v v17.5.0
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok
```

ajfisher, 2022-02-21:
This might be related to a slightly older version of serialport being used which may not be supported on a M1 given the architectural differences. This is rolled into both johnny-five and firmata which are both peer-dependencies so you don't have to use the version specific in the package.json

As such, try just installing `npm install firmata` and `npm install johnny-five` directly and see if that resolves.

closed 2022-03-05 by itsflores

ScreamZ, 2023-10-26:
is that the only way to solve this ?