ajfisher/node-pixel #371: Explanation on multiple stripe

ScreamZ, 2023-11-30:
Hello,

I've read that strip declared together are bound on a board.

What if I do this ?

```ts
 const strip1 = new LEDStrip({ board, strips:  [{ pin: 3, length: 196 }], gamma: 2.8 });
 const strip2 = new LEDStrip({ board, strips: [{ pin: 13, length: 12 }], gamma: 2.8 });
  ```

Are those strips not supposed to be separated ? It looks like not.
Do i have to rely on custom function to select the 12 last pixels on the second strip when playing with them ?

ajfisher, 2023-11-30:
Honestly, I'm not exactly certain what will happen in this scenario. 🤔

I think the creation of the second strip will probably redefine the first
on the firmware side. If you have the strips and the board available then
give it a go and report back what happens.

I don't think I considered this set up when I first designed it but it's an
interesting use case. I'm assuming you're using the two strips to drive
different lighting displays?

ajfisher, 2023-12-01:
> I'm working on an escape room project and I have on one board two
> different strips.
>
>    1.
>
>    The first one is used to represent a matrix of LEDs with color
>    indicating the riddle to solve. But the room is quite dark, that's why I
>    have a second strip.
>    2.
>
>    The second strip is used with 12 LEDs with maximum white color to
>    power a « Light » that is lighting the panel.
>
> ------------------------------
> After the first test
>
> I can confirm that strange things happen in this scenario. Like there are
> side effect when playing with one strip that appears on the other. It's
> like using the first one impact the secondly declared, but I'm not sure
> because I quickly switch to some kind of wrapper, that returns function
> simulating "fake strips" bound to only the LEDs they are supposed to acting
> on.
>
> Even though this is quite hacky, I think we should imagine some way to
> design this (even if we use a wrapper) to properly address this.
>
> In my case the first strip is about 196 LEDs, the second about 12. Does
> the limitation of LEDs apply on both strip total led count of its a per
> strip limitation. I guess its on a total one isn't ?

Cool project!

So yes, total LED count will be a hard limit because of the memory
limitations on the arduino. YOu can switch to a mega which is has a LOT
more memory and your totals become a couple of thousand.... Might be worth
dropping your total on the matrix to like 14 so it's just using one row and
seeing if that works. Then you can rule out memory issues.

In terms of the firmware, this shouldn't be the end of the world because
even in the multistrip scenario, it's configuring multiple strips and
arrays in the background as it needs to control the flow of data onto each
of the pins.

If the test above doesn't work (in which case it's a memory issue) then the
hacky solution to get you going is define them as running onto each other
and then in your application make a distinction about writing to the lower
196 (ie matrix display) v upper 12 (lighting) so you're keeping them
separate. Are you doing anything else with those lighting LEDs or it's just
for panel lighting? (If so, a simple white LED strip might be a simple
option as you just need to power it on or off via a pin then as a switch...)

The slightly more complex solution is we'd need to make a mod to the
protocol to allow a different configuration of strips so it's not treating
them like a group and you're maintaining them independently...